qaul/qaul.net #77: Command-line version?

deavmi, 2015-10-10:
Is there a way to use qaul.net in the command-line?

MathJud, 2015-10-17:
Not totally.
In the old qaul.net there is a description in how to run a headless qaul.net binary:
https://github.com/WachterJud/qaul.net/tree/master/cli

qaul.net-ng shall be able to start a headless qaul.net server and configure all important options in milestone beta5
issues: #65 

There is no milestone set for a CLI UI of qaul.net ( issue #78 )

MathJud, 2016-09-18:
qaul daemon can be run from the command line using

    qauld

To configure qaul check the documentation: 

    https://github.com/WachterJud/qaul.net/blob/qaul-ng/doc/CLI-Client_Configuration.md

closed 2016-09-18 by MathJud

deavmi, 2016-09-19:
Awesome :

deavmi, 2016-09-19:
Awesome :+1: Thanks :)