qaul/qaul.net #82: Arch Linux - Build Error(s)

g4jc, 2015-11-07:
Hello,
I stumbled across your program and decided to try compiling it. Unfortunately, it fails.
I have fixed two of the linking errors which are caused by missing lines in /qaul.net-ng/src/libqaul/CMakeLists.txt. The patches are for pthreads and uuid fixes respectively. Available here: http://dpaste.com/328PKSY.txt

However, once that patch is applied it fails again, and is a bit beyond my scope of understanding.
'''
[100%] Linking C executable qaul-gtk
../../../third_party/pjsip/src/pjsip/pjmedia/lib/libpjmedia-codec-qaul.a(opencore_amr.o): In function `amr_codec_decode':
opencore_amr.c:(.text+0x3d0): undefined reference to `Decoder_Interface_Decode'
../../../third_party/pjsip/src/pjsip/pjmedia/lib/libpjmedia-codec-qaul.a(opencore_amr.o): In function `amr_codec_encode':
opencore_amr.c:(.text+0x652): undefined reference to `Encoder_Interface_Encode'
../../../third_party/pjsip/src/pjsip/pjmedia/lib/libpjmedia-codec-qaul.a(opencore_amr.o): In function `amr_codec_close':
opencore_amr.c:(.text+0xf39): undefined reference to `Decoder_Interface_exit'
opencore_amr.c:(.text+0xf41): undefined reference to `Encoder_Interface_exit'
../../../third_party/pjsip/src/pjsip/pjmedia/lib/libpjmedia-codec-qaul.a(opencore_amr.o): In function `amr_codec_modify':
opencore_amr.c:(.text+0xfcb): undefined reference to `Encoder_Interface_exit'
opencore_amr.c:(.text+0xfd3): undefined reference to `Encoder_Interface_init'
../../../third_party/pjsip/src/pjsip/pjmedia/lib/libpjmedia-codec-qaul.a(opencore_amr.o): In function `amr_codec_open':
opencore_amr.c:(.text+0x13d1): undefined reference to `Encoder_Interface_init'
opencore_amr.c:(.text+0x13f4): undefined reference to `Decoder_Interface_init'
collect2: error: ld returned 1 exit status
src/client/gtk/CMakeFiles/qaul-gtk.dir/build.make:140: recipe for target 'src/client/gtk/qaul-gtk' failed
make[2]: *** [src/client/gtk/qaul-gtk] Error 1
CMakeFiles/Makefile2:427: recipe for target 'src/client/gtk/CMakeFiles/qaul-gtk.dir/all' failed
make[1]: *** [src/client/gtk/CMakeFiles/qaul-gtk.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
'''

MathJud, 2015-11-11:
Hi!
We have qaul.net running on Arch machines.
Unfortunately we couldn't reproduce your Error.
You should be able to build qaul.net without the patch you attached.

Have checked out the most recent git version?
On what platform are you building?

g4jc, 2015-11-12:
Hello!
Thanks for the response, I am using the latest qaul.net-ng git clone.
Currently I'm attempting to build on on a standard x86_64 PC with cmake-3.3.2, gcc-5.20 and GNU/Linux kernel 4.2.5.

My PKGBUILD for compiling and packaging it using makepkg currently looks like this:
http://pastebin.com/raw.php?i=9C84wCnf

closed 2015-11-12 by MathJud

g4jc, 2015-11-13:
Why was this closed? It's not resolved yet.

I should have stated my "non-working" PKGBUILD. If you want I can keep patching the software, but it seems to be bugs with the latest cmake and dependencies.