qaul/qaul.net #85: Switch to batman-adv

deavmi, 2016-01-16:
Batman-adv is way better than olsrd. And has reactive routing which is way better in performance and reliability.

MathJud, 2016-01-19:
Unfortunately Batman-adv is only running on Linux. As qaul.net supports many OSs we can't use this protocol.

closed 2016-01-19 by MathJud

deavmi, 2016-01-20:
Are there no ports?

MathJud, 2016-01-20:
Unfortunately not.

deavmi, 2016-07-03:
What about the newer bmx6?

deavmi, 2016-07-06:
Have you looked into the following?:

* Babel: https://www.irif.univ-paris-diderot.fr/~jch/software/babel/
* bmx6: http://bmx6.net/projects/bmx6

Or cjdns? (Which I think could work on all Demario platforms (android possibly one day - somebody was working on an android version I think))

MathJud, 2016-07-13:
I know these protocols.
It would be cool to be able to use different protocols. For that, one would need an own mesh-enabled message floader. At the moment qaul.net uses the flooding system of olsrd.
 
cjdns is an Internet overlay network, it is not suitable for a quickly changeable mobile wifi mesh routing.

deavmi, 2016-07-13:
cjdns can be used in a non-overlay way it can do MAC-routing apparently.

deavmi, 2016-07-13:
Hopefully a build of batman-adv for Mac will happen soon. Is it suitable for mesh networks with nodes moving around and re-configuring a lot?

deavmi, 2016-07-13:
And Windows...