qaul/qaul.net #86: Do initial home-folder creation and copying in qaullib

MathJud, 2016-01-17:
When qaul starts for the first time, a home folder is created and the files folder is copied into this home folder.
At the moment this procedure is done by every client individually.
It would be great to have that implemented into qaullib for that every client could use the same mechanism. 

This would have the following advantages:
* there could be a central updating procedure due to the version of qaul.net
* client ports are easier because this does not have to be written for every client separately.

spacekookie, 2016-05-11:
I would suggest the home folder is moved to `~/.config/qaul[...]` on *nix systems. Most utilities and applications already put their configurations into `.config` as it helps to un-clutter users home directories that is usually very full of hidden folders or files for tiny utilities.

deavmi, 2016-07-07:
@spacekookie I agree.

closed 2019-03-26 by spacekookie