qaul/qaul.net #92: What discussion channels should be created for qaul.net

MathJud, 2016-02-02:
What possibilities should be created to discuss problems, configuration, concepts etc. of qaul.net.
* Mailing list
* Forum
* IRC
* Wiki
* Nothing (github is enough)

Thanks for your thoughts.

MathJud, 2016-02-02:
I would be in favor of a mailing list.

anuvazhayil, 2016-03-05:
Is this discussion still open? I have a suggestion to make. How about installing a Phabricator, it can track all tasks and activities. We can also create a thread for open discussion and good interface :D http://phabricator.org/

jaromil, 2016-03-20:
If needed, we can host the mailinglist on lists.dyne.org: we maintain that already, happy to share.

deavmi, 2016-07-07:
IRC.

deavmi, 2016-07-07:
I think IRC and mailing list (I like mailing list the most probably).

closed 2019-03-26 by spacekookie