qaul/qaul.net #99: qaullib log messages: create log function/macro

MathJud, 2016-02-17:
The following functionality should be reached:
* configure verbosity of feedback
* possibility to exclude messages from the binary when compiling. To receive a smaller binary when compiling for embedded systems.

MathJud, 2016-05-25:
The logging macro exists ().
However there are still the following issues when using it:
* Windows build fails

The following test file can be used to reproduce the result (uncomment the Ql_log_... functions and the header file): 
https://github.com/WachterJud/qaul.net/blob/qaul-ng/src/libqaul/qaullib_crypto.c 

The Windows build error message:
```
CMakeFiles/libqaul.dir/objects.a(qaullib_crypto.c.obj):qaullib_crypto.c:(.data+0x1e10): multiple definition of `loglevel'
CMakeFiles/libqaul.dir/objects.a(logging.c.obj):logging.c:(.data+0x0): first defined here
CMakeFiles/libqaul.dir/objects.a(qaullib_crypto.c.obj):qaullib_crypto.c:(.rdata+0x10a8): multiple definition of `LOG_LEVEL_NAMES'
CMakeFiles/libqaul.dir/objects.a(logging.c.obj):logging.c:(.rdata+0x1c): first defined here
collect2.exe: error: ld returned 1 exit status
src/libqaul/CMakeFiles/libqaul.dir/build.make:865: die Regel für Ziel „src/libqaul/libqaul.dll“ scheiterte
make[2]: Verzeichnis „/home/vostok/builds/c20480df/0/WachterJud/qaul.net/build“ wird verlassen
CMakeFiles/Makefile2:278: die Regel für Ziel „src/libqaul/CMakeFiles/libqaul.dir/all“ scheiterte
make[1]: Verzeichnis „/home/vostok/builds/c20480df/0/WachterJud/qaul.net/build“ wird verlassen
make[2]: *** [src/libqaul/libqaul.dll] Fehler 1
make[1]: *** [src/libqaul/CMakeFiles/libqaul.dir/all] Fehler 2
Makefile:149: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2
```

MathJud, 2016-05-27:
The logging macro is now functional.
* Default log level can be set in cmake -DQAUL_LOG_DEFAULTLEVEL=[ NONE | ERROR | WARN | INFO | DEBUG]

The following functionality is still missing:
* Cmake option -DQAUL_LOG_ENABLE=NO
* change log level when starting the program.

MathJud, 2016-06-01:
Logger build notes and warnings on ubuntu 14.04:

```
[ 74%] Building C object src/libqaul/CMakeFiles/libqaul.dir/qaullib_crypto.c.o
In file included from /home/sojus/src/meshnet/qaul.net/src/libqaul/qaullib_crypto.c:9:0:
/home/sojus/src/meshnet/qaul.net/src/libqaul/qaullib_crypto.c: In function ‘Ql_sha1_file’:
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:139:9: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘__suseconds_t’ [-Wformat=]
  struct timeval tv; \
         ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:92:30: note: in expansion of macro ‘Ql_logline’
 #define Ql_log_debug(M, ...) Ql_logline(LOG_DEBUG, 0, M, ##__VA_ARGS__)
                              ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/qaullib_crypto.c:21:5: note: in expansion of macro ‘Ql_log_debug’
     Ql_log_debug("Ql_sha1_file %s", filepath);
     ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:139:9: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘__suseconds_t’ [-Wformat=]
  struct timeval tv; \
         ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:92:30: note: in expansion of macro ‘Ql_logline’
 #define Ql_log_debug(M, ...) Ql_logline(LOG_DEBUG, 0, M, ##__VA_ARGS__)
                              ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/qaullib_crypto.c:21:5: note: in expansion of macro ‘Ql_log_debug’
     Ql_log_debug("Ql_sha1_file %s", filepath);
     ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:139:9: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘__suseconds_t’ [-Wformat=]
  struct timeval tv; \
         ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:106:59: note: in expansion of macro ‘Ql_logline’
 #define Ql_log_error(M, ...) if ( LOG_ERROR <= loglevel ) Ql_logline(LOG_ERROR, 1, M , ##__VA_ARGS__)
                                                           ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/qaullib_crypto.c:26:6: note: in expansion of macro ‘Ql_log_error’
      Ql_log_error("Ql_sha1_file failed to open file: %s\n", filepath);
      ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:139:9: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘__suseconds_t’ [-Wformat=]
  struct timeval tv; \
         ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:106:59: note: in expansion of macro ‘Ql_logline’
 #define Ql_log_error(M, ...) if ( LOG_ERROR <= loglevel ) Ql_logline(LOG_ERROR, 1, M , ##__VA_ARGS__)
                                                           ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/qaullib_crypto.c:26:6: note: in expansion of macro ‘Ql_log_error’
      Ql_log_error("Ql_sha1_file failed to open file: %s\n", filepath);
      ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:139:9: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘__suseconds_t’ [-Wformat=]
  struct timeval tv; \
         ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:106:59: note: in expansion of macro ‘Ql_logline’
 #define Ql_log_error(M, ...) if ( LOG_ERROR <= loglevel ) Ql_logline(LOG_ERROR, 1, M , ##__VA_ARGS__)
                                                           ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/qaullib_crypto.c:40:6: note: in expansion of macro ‘Ql_log_error’
      Ql_log_error("Ql_sha1_file file error on file: %s\n", filepath);
      ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:139:9: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘__suseconds_t’ [-Wformat=]
  struct timeval tv; \
         ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:106:59: note: in expansion of macro ‘Ql_logline’
 #define Ql_log_error(M, ...) if ( LOG_ERROR <= loglevel ) Ql_logline(LOG_ERROR, 1, M , ##__VA_ARGS__)
                                                           ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/qaullib_crypto.c:40:6: note: in expansion of macro ‘Ql_log_error’
      Ql_log_error("Ql_sha1_file file error on file: %s\n", filepath);
      ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/qaullib_crypto.c: In function ‘Ql_HashToString’:
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:139:9: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘__suseconds_t’ [-Wformat=]
  struct timeval tv; \
         ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:92:30: note: in expansion of macro ‘Ql_logline’
 #define Ql_log_debug(M, ...) Ql_logline(LOG_DEBUG, 0, M, ##__VA_ARGS__)
                              ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/qaullib_crypto.c:59:2: note: in expansion of macro ‘Ql_log_debug’
  Ql_log_debug("Ql_HashToString");
  ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:139:9: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘__suseconds_t’ [-Wformat=]
  struct timeval tv; \
         ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/include/qaullib/logging.h:92:30: note: in expansion of macro ‘Ql_logline’
 #define Ql_log_debug(M, ...) Ql_logline(LOG_DEBUG, 0, M, ##__VA_ARGS__)
                              ^
/home/sojus/src/meshnet/qaul.net/src/libqaul/qaullib_crypto.c:59:2: note: in expansion of macro ‘Ql_log_debug’
  Ql_log_debug("Ql_HashToString");
  ^
```

deavmi, 2016-07-07:
a function that takes in a `char[] logMessage`?

deavmi, 2016-07-07:
As in a charcater array

deavmi, 2016-07-07:
Oh I see this is implemented already. Nevrmind then. :)

closed 2019-03-20 by spacekookie