qaul/qaul.net #104: Generate the concept for a sharable wiki / web pages

MathJud, 2016-03-12:
Generate the concept for a sharable wiki / web pages
* Synchronizable and sharable wiki pages shall be part of qaul.net.
* They shall be visible for every one in the network.
* They shall be editable (locally and remote)
* They shall be sharable among users and be stored locally, with the possibility
  of synchronization but also differation (similar to a git repository)
* They shall be able to contain media objects, such as images & videos.

jaromil, 2016-03-20:
How about running gobby?
https://gobby.github.io/

jaromil, 2016-03-20:
or build something with their https://github.com/gobby/libinfinity

deavmi, 2016-07-06:
Yeah, just run a service. And it should be available on your IP at a given port on the mesh.

MathJud, 2016-07-06:
libinifity looks really interesting, but I think we should create an entirely decentralized solution.

deavmi, 2016-07-06:
I truly am all about just the mesh. I believe people should decide **what** services run on the mesh and **how**. My humble opinion.

deavmi, 2016-07-06:
What is libinfinity?

MathJud, 2016-07-13:
@deavmi check link in jaromil's post to libinfinity  https://github.com/gobby/libinfinity

deavmi, 2016-07-13:
I see. I guess Etherpad would have some similiar system.

jaromil, 2016-07-24:
I generally agree with @deavmi on his stance to focus on the mesh. If you keep it simple and on the low layer, it will all keep more maintainable and leave you more time to work on details, while leaving people the freedom to run any stuff on application space. Its a good mantra for the programmer, to not build-in features, but leave space for inter-operability. Also process separation helps stability.

deavmi, 2016-07-24:
Good words in last comment

closed 2019-03-26 by spacekookie