qaul/qaul.net #107: IP Area /16 inside freifunk 10/8

FreifunkUFO, 2016-05-07:
the generation of ip-adresses is too randomly atm. its not ok to use a very random ip from whole networkarea of an town.
better use one 10.x/16 which is reserved inside freifunk tables (ic-vpn) for all qaul nodes.
presets are still needed, f.i. channel settings

https://github.com/WachterJud/qaul.net/blob/master/GUI/www/qaul.js

FreifunkUFO, 2016-05-23:
any ideas?

MathJud, 2016-05-24:
In the community template for Leipzig, there is a 16bit mask set for 10.61.x/16:
https://github.com/WachterJud/qaul.net/blob/master/GUI/www/community_templates/ff_leipzig

So whenever you use the community template for freifunk it will be configured just fine.
Where is the problem to use the whole 10.x/24 space for qaul.net?

FreifunkUFO, 2016-05-25:
vielleicht diskutieren wir lieber in deutsch :-)
bitte nehmt NICHT eine random-IP aus dem 10.x/16-Bereich, dies gilt zumindest für Leipzig.
wir hatten uns vor einigen Jahren mal auf einen eigenen 10.x/16 bereich für qual.net geeinigt, egal in welcher Stadt der qual-router steht. als presets wären hier dann nur die wlan-parameter je stadt zu ändern. warum habt ihr das mittlerweile anders gemacht?

zu deiner o.g. frage gleich eine gegenfrage: wie berechnest du aus 10.61.x/16 und einer 16er Maske ein 10.x/24? bzw welches /24 Netz meinst du dann damit?

MathJud, 2016-05-25:
Kein Problem!
Wir können im community-template einfach die auto-generation ausschalten, dann muss man selbst eine IP-Adresse eingeben. Nur die default werte sind dann schon sinnvoll gesetzt.
Wir setzen einfach den key 'generate' von true auf false:

  "ip":{"edit":"true","generate":"false","pattern":"10.61.0.0/16"},

Wäre dies dann OK? Oder gibt es noch andere Fehler?

Im Feld Info können wichtige Informationen eingetragen werden. z.B.:
Bitte registriert zuerst eine IP-Adresse unter http://XXXXX und gebt diese dann im Feld IP ein.

  "profile":{"name":"Freifunk Leipzig","homepage":"http://leipzig.freifunk.net","info":"Bitte registriert zuerst eine IP-Adresse unter http://XXXXX und gebt diese dann im Feld IP ein."},

Wo kann man denn ohne Firmware eine IP-Adresse registrieren?

FreifunkUFO, 2016-05-26:
Die Registrierung erfolgt bei uns ggf. im wiki. In Berlin klickt man durch einen Online-Wizard und kann sich unter Eingabe der email-adresse IPs zuweisen lassen (ggf. muss man bei "verwendetes Gerät" schummeln.)

Ja, leider wäre es das beste, derzeit die auto-generation komplett abzuschalten :-(
Bitte erkläre mal die Umrechnung von 2x /16 Subnetz in ein /24! 
Warum wollt ihr kein bundesweit einheitliches 10.x/16  Subnetz für qual nehmen wie vor 2 Jahren besprochen?

Was meinst du mit default werten? die Grundeinstellungen für wlan-channel und bssid?

MathJud, 2016-06-01:
Ich werde den Link ins Wiki bei Leipzig-Community-Template setzen und die autogeneration abschalten.
Ist dies die korrekte Seite? http://wiki.leipzig.freifunk.net/Bereiche

Einen Bereich für qaul.net für einen allgemeines Freifunk-Community-Template mit auto generation der Adresse finde ich gut, wo müsste der eingetragen werden?
Als default für qaul.net wäre es weniger günstig, da die Gefahr von Kollisionen zunimmt und es nur in Deutschland sinn macht.

closed 2019-03-26 by spacekookie

FreifunkUFO, 2019-03-29:
ok @spacekookie , did you recently change your qaul-setup?

spacekookie, 2019-03-29:
Yea, we're currently in the process of rebooting the project in a few ways

FreifunkUFO, 2019-04-01:
ah ok.. 

took me some time, but i found some information about that: https://blog.freifunk.net/2018/05/18/designing-a-userspace-routing-protocol-for-qaul-net/

so qual doesnt use OLSR anymore.. (p.s. i hope that ratman is compatible to batman-adv)

spacekookie, 2019-04-01:
`ratman` runs on a higher layer than `BATMAN` or `OLSR`, i.e. it's entirely possible to write a backplane adapter for both of them and plug into existing Freifunk networks.

This isn't our highest priority but definitely on the roadmap (and technically not blocked)