qaul/qaul.net #108: OSX El Capitan cannot download the installer from the captive portal.

MathJud, 2016-06-24:
Apple changed the behaviour of it's login browser with captive portals. 
The login browser opens the download link with the default web browser. 
The web browser is then told by the system that no network connection exists.

Ideas and solutions are welcome

deavmi, 2016-07-06:
Oh. Does it have a problem checking for an Internet connection? And that wouldn't work on an offline mesh net?

MathJud, 2016-07-13:
Someone told me that it can be configured on a per network basis in OSX and iOS.
If this works, we should show implement an screenshot based explanation for those systems how a user can do that.

deavmi, 2016-07-13:
Sounds good @MathJud :+1:

MathJud, 2016-09-18:
OSX checks for the URL "http://captive.apple.com/hotspot-detect.html" and needs to be returned the following HTML:

    <HTML><HEAD><TITLE>Success</TITLE></HEAD><BODY>Success</BODY></HTML>

commit 0da959c4 (2016-10-10): OSX: Captive Portal Checker: Added web server response, fixes #108
closed 2016-10-10 by MathJud