qaul/qaul.net #112: New nightlies

deavmi, 2016-07-03:
By the way, I tested out the new nightlies, they are awesome.

However, the Windows version seems to be behind in features and did crash a few times on initial start-up. I installed many versions before the latest to see which ones worked better but yeah.

I tested out the Internet sharing on my Linux laptop with the Windows desktop and it worked (even YouTube - not the best performance obviously).

MathJud, 2016-07-05:
Thanks for the testing!
Every time a new commit is published on branch qaul-ng the buildserver automatically generates a new nightly binary. 
(Regardless whether it was typo fix or a big change in functionality.)
The installers have the ID of the related git commit in their name, for that one can check the differences.

> However, the Windows version seems to be behind in features and did crash a few times on initial start-up. I installed many versions before the latest to see which ones worked better but yeah.
> 

Thanks for the hint about the crashes!!! We have to look into that!!!!

Unfortunately Windows is behind the following features:
* The Internet sharing button is missing. 
  We lack a dynamic gateway plugin for OLSR on Windows that could deside whether or not Internet is really present. 
* The user configuration and the users files are put in the installation folder, the shared files included. 
  As in all the other distros we should copy the 
* The whole windows application is running with administrative rights. 
  All other platforms run the application in user space with a special helper binary with suid rights.

It would be really great to have more windows developers and to change that!!!

> I tested out the Internet sharing on my Linux laptop with the Windows desktop and it worked (even YouTube - not the best performance obviously).
> 
Yey!!!

deavmi, 2016-07-05:
Awesome. The Internet connection option worked on Linux and I was able to get Internet via my Linux machine to my Windows machine.

deavmi, 2016-07-05:
Love the addition of a build sever and beautiful page for binaries to the project. Really awesome when tests are needed to be done, makes it easy for the end user.

deavmi, 2016-07-06:
When I try to visit the "Show Network" page it crashes qaul.net in the Windows version.  Also the "Show network" button does nothing in the Windows version.

deavmi, 2016-07-06:
My Windows version feels very broke. Going to try install an older beta version.

deavmi, 2016-07-06:
Just tested running Boa web server over aul.net and it worked.

closed 2019-03-26 by spacekookie