qaul/qaul.net #140: Headless dedicated node builds

deavmi, 2016-07-07:
I think it would be really awesome to see headless builds with no voip, chat or file sharing but with the easy to setup Qaul network system. 

As then people an easily install it on a Raspberry Pi running Raspbian in console mode and create a real dedicated node.

These builds could also disable the Qaul web interface but keep the show network interface (i don't know, maybe - maybe not).

If you see where I am going.

MathJud, 2016-07-10:
Yep!!!
Our plans are to create Raspber-pi and openWRT/LEDE images to easily create seeding nodes as well as infrastructure nodes.

deavmi, 2016-07-10:
Awesome.

deavmi, 2016-07-10:
Seeding nodes?

deavmi, 2016-07-10:
Also. Take a look at the Babel routing protocol. There is a blog post on how to set it up. http://www.ericerfanian.com/mobile-mesh-networks-with-the-raspberry-pi-part-3/

Maybe we could attempt integrating it into Qaul. Need to see if it will build on Windows and MacOSX tho.

deavmi, 2016-07-10:
Also, will the CLI builds be able to start olsr router soon? If I am correct they don't do any network configuration yet?

MathJud, 2016-09-18:
qauld can be run headless on  raspbian.
The configuration can be done directly in the data base following the documentation: 
https://github.com/WachterJud/qaul.net/blob/qaul-ng/doc/CLI-Client_Configuration.md

It should become easier once issue #65  is solved.

closed 2016-09-18 by MathJud