qaul/qaul.net #149: No CSS?

deavmi, 2016-07-12:
Ran the latest version of `qaul-gtk` from the terminal and when viewing the pages there seemed to be a lack of styling applied to them?

[image]

deavmi, 2016-07-12:
This page didn't seem to be working all that well either.

[image]

It didn't show any nodes (it should of shown one - myself)

MathJud, 2016-07-13:
As your screenshot quotes, you might have to enable java-script in your browser to see the nodes (and most probably also the index page).

deavmi, 2016-07-13:
I have enabled it though. Let me check again.

deavmi, 2016-07-13:
I am very sure my javascript is enabled.

deavmi, 2016-07-13:
Yeah, it is. So sure.

MathJud, 2016-09-18:
I can't verify this issue. Does it still exist?
The CSS file (qaul_web.css) is in all the install packages. 
It is located under: /usr/local/lib/qaul/www/qaul_web.css

closed 2016-09-18 by MathJud

deavmi, 2016-09-19:
I will test this again.