qaul/qaul.net #150: Add GUI path configuration option

MathJud, 2016-07-13:
The path to the GUI is hard coded in qaul.net.
The path should be configurable for the following reasons:
* to not to have to install the interface every time something has changed during the GUI development
* to allow the user to use alternative home-folder located GUI's

deavmi, 2016-07-13:
What do you mean by this? How is the path to the GUI hard-coded? What the `qaul-gtk`?

MathJud, 2016-07-13:
qaullib locates the GUI path automatically within the path that set in the Qaullib_Init function which is invoked by the client. It should get the path from the config table of the data base.

deavmi, 2016-07-13:
Ever thought of switching to a config file for some aspects of the qaul.net system? I'd love this.

The database system is good for large data or multiple pieces of data being stored.

closed 2019-03-20 by spacekookie