qaul/qaul.net #159: can't find use internet connection

netachepas, 2016-09-12:
I just tried to launch qaul-gtk on my laptop, running devuan gnu/linux and try to share internet connection but it does not seem to find the network interface.... it says

```
Qaullib_Init:
resource path: /usr/local/lib/qaul
home path: /home/natacha/.qaul
Qaullib_WwwWebGetMsgs
Qaullib_MsgInit
Qaullib_MsgDB2LL
Qaullib_FileInit
Qaullib_FileDB2LL
Qaullib_ExeInit
Qaullib_ExeInit: exe.4 has not been discovered
Qaullib_ExeInit: exe.8 has not been discovered
Qaullib_ExeInit: exe.16 has not been discovered
Qaullib_ExeInit: exe.64 has not been discovered
Qaullib_ExeInit: exe.128 has not been discovered
Qaullib_ExeInit: qaul_exe_available 0
Qaullib_WebserverStart
Qaullib_WgetRunThread create type: 2
Qaullib_WgetRunThread create type: 2
Qaullib_WgetRunThread create type: 1
Qaullib_WgetRunThread create type: 1
Qaullib_WgetRunThread create type: 2
Qaullib_WgetRunThread create type: 2
Qaullib_WgetRunThread create type: 2
Qaullib_WgetRunThread create type: 1
qaul_web_localip conn->sa.sin.sin_addr 127.0.0.1
qaul_network_find_wifi
qaul_network_find_wifi dbus_message_iter_init | DBUS_TYPE_ARRAY error
[configure] No WiFi interface found
[configure] start olsrd
start olsrd
missing argument
[configure] connect ipc
Error connecting 111 - Connection refused
[qaullib] IPC: Attempting connect...[configure] start captive portal
17:47:39.643 os_core_unix.c !pjlib 2.5 for POSIX initialized
17:47:39.644 sip_endpoint.c  .Creating endpoint instance...
17:47:39.644          pjlib  .select() I/O Queue created (0x166e0f0)
17:47:39.644 sip_endpoint.c  .Module "mod-msg-print" registered
17:47:39.644 sip_transport.  .Transport manager created.
17:47:39.644   pjsua_core.c  .PJSUA state changed: NULL --> CREATED
17:47:39.644 sip_endpoint.c  .Module "mod-pjsua-log" registered
17:47:39.644 sip_endpoint.c  .Module "mod-tsx-layer" registered
17:47:39.644 sip_endpoint.c  .Module "mod-stateful-util" registered
17:47:39.644 sip_endpoint.c  .Module "mod-ua" registered
17:47:39.645 sip_endpoint.c  .Module "mod-100rel" registered
17:47:39.645 sip_endpoint.c  .Module "mod-pjsua" registered
17:47:39.645 sip_endpoint.c  .Module "mod-invite" registered
Home directory not accessible: Permission denied
Home directory not accessible: Permission denied
Home directory not accessible: Permission denied
Home directory not accessible: Permission denied
17:47:39.802     alsa_dev.c  ..ALSA driver found 16 devices
17:47:39.802     alsa_dev.c  ..ALSA initialized
17:47:39.802          pjlib  ..select() I/O Queue created (0x17274e8)
WARNING: no real random source present!
17:47:39.807 sip_endpoint.c  .Module "mod-evsub" registered
17:47:39.807 sip_endpoint.c  .Module "mod-presence" registered
17:47:39.807 sip_endpoint.c  .Module "mod-mwi" registered
17:47:39.807 sip_endpoint.c  .Module "mod-refer" registered
17:47:39.807 sip_endpoint.c  .Module "mod-pjsua-pres" registered
17:47:39.807 sip_endpoint.c  .Module "mod-pjsua-im" registered
17:47:39.807 sip_endpoint.c  .Module "mod-pjsua-options" registered
17:47:39.807   pjsua_core.c  .1 SIP worker threads created
17:47:39.807   pjsua_core.c  .pjsua version 2.5 for Linux-3.16.0.4/x86_64/glibc-2.19 initialized
17:47:39.807   pjsua_core.c  .PJSUA state changed: CREATED --> INIT
17:47:39.807   pjsua_core.c  SIP UDP socket reachable at 192.168.178.185:8060
17:47:39.807   udp0x1701980  SIP UDP transport started, published address is 192.168.178.185:8060
17:47:39.807   pjsua_core.c  PJSUA state changed: INIT --> STARTING
17:47:39.807 sip_endpoint.c  .Module "mod-unsolicited-mwi" registered
17:47:39.807   pjsua_core.c  .PJSUA state changed: STARTING --> RUNNING
17:47:39.808    pjsua_acc.c  Adding account: id=<sip:192.168.178.185:8060>
17:47:39.808    pjsua_acc.c  .Account <sip:192.168.178.185:8060> added with id 0
17:47:39.808    pjsua_acc.c  Modifying account 0
UDP server started
Qaullib_CaptiveStart
start portforwarding
missing argument
[configure] timers
[configure] Finished
...........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................[qaullib] IPC connection lost!
Qaullib_ExeScheduleDiscovery send flag 220
[qaullib] IPC connection lost!
```

Any info welcome

MathJud, 2016-09-15:
When you tested this, the Linux network configuration was based on network-manager. As you don't have network-manager installed on your system the network configuration failed.

However, we added a new configuration method which is automatically used when network-manager isn't present. The new configuration method configures your system using the CLI tools 'ip', 'iwconfig', 'iw', ...
You may build the current development branch 'qaul-ng' and try it on your system. It may still have flaws. I'm interested in your feedback!

MathJud, 2016-09-15:
related to #96

closed 2016-09-15 by MathJud

netachepas, 2016-09-15:
Thanks loads MathJud,

this seems to share connection easily did not try yet to use it on
several devices, will keep you updated.

Also I am thinking of qaul for q series of workshop that I am suppose to
give in a Brussel based organisation Zinneke, www.zinneke.org.

Its quite a big biennal parade in Brussels that is structured around the
participation of a large number of community based structures, their
each time choose a topic next will be legal/illegal and they asked me if
I could present a project that would help include digital processes both
in their preliminary workflow and maybe during the parade itself.

my idea was to propose a series of workshops, before the parade so that
each community organisation taking part in the process of the
organisation of the parade would be able to set up a qaul network within
their group and then provide them with a platform to share their images
and feelings during the parade itself.

I do not yet have much practical information as I am only starting the
conversation with the organisation, but I think it might be a good
opportunity to access a broad public live experiments.

It would be great to talk about it with you

regards

natacha

On 09/15/2016 08:08 AM, MathJud wrote:
> When you tested this, the Linux network configuration was based on network-manager. As you don't have network-manager installed on your system the network configuration failed.
>
> However, we added a [...]