qaul/qaul.net #177: Create user store API

spacekookie, 2017-08-04:
Define an API that will be used to interact with the user store, creating an abstraction layer around the active user store (hash tables) and the all-known-store (database)

MathJud, 2017-08-07:
# General Concept
We should separate network view from user store, with the possibility to map multiple users to the same Network node.

# User Store
I would prefer a user store internal structure, as it could be more dynamic.
We should only have some very basic information as static fields.
While we should be able to find the user entry via IP and fingerprint (maybe even via username), user edits (delte, change, etc.) could be done only via fingerprint.

e.g.
* Fingerprint
* Username
* additional fields:
  * public key
  * trust fields

# Network structure
The network structure should be extendable to multiple network systems.
The first system to implement should be the IPv4 network (which is used in the IBSS olsrd network).

To be able to interconnect the same node over different networks, we should maybe think about giving the node an ID to which is independent of the IP address.

spacekookie, 2017-08-31:
Right now, a user is mapped to a network node which is defined as follows

```C
/** Holds data about a node */
typedef struct qluser_node_t {
    union olsr_ip_addr  *ip;
    // ...
} qluser_node_t;
```

We probably want to create an abstraction so there is no direct mention of olsr outside of any code that specifically uses olsr. But that's how it works for now. Multiple users can be on the same node and you can also quickly find all users that use a specific IP address

closed 2019-03-20 by spacekookie