qaul/qaul.net #194: Fails to build on Arch Linux

spacekookie, 2017-10-14:
```
[ 94%] Built target qauld
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.0/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `cairo_gl_surface_get_width'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.0/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `cairo_egl_device_create'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.0/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `cairo_glx_device_create'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.0/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `cairo_gl_surface_get_height'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.0/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `cairo_gl_device_set_thread_aware'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.0/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `cairo_gl_surface_create_for_texture'
collect2: error: ld returned 1 exit status
make[2]: *** [src/client/gtk/CMakeFiles/qaul-gtk.dir/build.make:248: src/client/gtk/qaul-gtk] Error 1
make[1]: *** [CMakeFiles/Makefile2:1844: src/client/gtk/CMakeFiles/qaul-gtk.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

@saces any ideas? :scream:

spacekookie, 2017-11-13:
Hmm, this isn't seem to be an issue anymore

spacekookie, 2018-01-04:
This has become an issue again, I think with the same error:

Log: https://pastebin.com/fGcPGyQZ (or below)

```
/usr/sbin/ld: warning: libenchant.so.1, needed by /usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so, not found (try using -rpath or -rpath-link)
/usr/sbin/ld: warning: libicui18n.so.59, needed by /usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so, not found (try using -rpath or -rpath-link)
/usr/sbin/ld: warning: libicuuc.so.59, needed by /usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so, not found (try using -rpath or -rpath-link)
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `cairo_gl_surface_get_width'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `usearch_setText_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `cairo_gl_surface_get_height'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libjavascriptcoregtk-3.0.so: undefined reference to `u_strFoldCase_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `cairo_gl_device_set_thread_aware'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libjavascriptcoregtk-3.0.so: undefined reference to `udat_format_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `ucnv_getAlias_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libjavascriptcoregtk-3.0.so: undefined reference to `ucol_strcoll_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `ucnv_reset_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `ublock_getCode_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `u_toupper_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `usearch_setOffset_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `ucnv_setToUCallBack_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `cairo_egl_device_create'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `enchant_dict_suggest'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libjavascriptcoregtk-3.0.so: undefined reference to `ucol_getLocaleByType_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `UCNV_FROM_U_CALLBACK_SUBSTITUTE_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `u_charMirror_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `ucnv_setFallback_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `vtable for icu_59::UnicodeString'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `ucnv_setFromUCallBack_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `uscript_getScript_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `enchant_dict_free_suggestions'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `ucnv_setSubstChars_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `u_isprint_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `usearch_next_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `ucnv_countAvailable_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `icu_59::UnicodeString::UnicodeString(char16_t const*, int)'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `ubrk_open_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `ucnv_cbFromUWriteUChars_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `ucnv_fromUnicode_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `uset_close_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libjavascriptcoregtk-3.0.so: undefined reference to `udat_open_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `uset_openPattern_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `enchant_broker_free_dict'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `unorm_quickCheck_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `cairo_gl_surface_create_for_texture'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `ucol_setStrength_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `enchant_dict_check'
[... cut ...]
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `ubrk_first_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `UCNV_FROM_U_CALLBACK_ESCAPE_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `usearch_setPattern_59'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/../../../../lib/libwebkitgtk-3.0.so: undefined reference to `ubrk_getRuleStatus_59'
collect2: error: ld returned 1 exit status
```

EkriirkE, 2018-11-07:
Same issue here as well, 
libjavascriptcoregtk-3.0.so expects libicui18n.so.59 libicuuc.so.59 but current version provided by core/icu is 63.1
libwebkitgtk-3.0.so expects libenchant.so.1 but current version supplied by extra/enchant is ibenchant-2.so.2.2.3

closed 2019-03-20 by spacekookie