qaul/qaul.net #196: Create File Structure for 2.0

MathJud, 2017-11-11:
Group the qaul src files into folders according to the modules they belong to in qaul 2.0

spacekookie, 2017-11-13:
## Proposed code file structure

```
libqaul/
├── api
│   ├── admin
│   │   ├── configuration.c
│   │   └── identities.c
│   ├── files
│   │   ├── receive.c
│   │   └── send.c
│   ├── messaging
│   │   ├── receive.c
│   │   └── send.c
│   ├── networking
│   │   ├── diagnostics.c
│   │   └── management.c
│   ├── qapi.c
│   ├── users
│   │   └── users.c
│   └── voip
│       └── calls.c
├── authentication
│   ├── qlauth.h
│   └── something.c
├── crypto-core
│   ├── qlcry.h
│   └── something.c
├── crypto-store
│   ├── qlcry_store.h
│   └── something.c
├── database
│   ├── qldb.h
│   └── something.c
├── filesystem
│   ├── qlfs.h
│   └── something.c
├── formatting
│   ├── qlform.h
│   └── something.c
├── networking
│   ├── qlnet.h
│   └── something.c
└── protocol
    ├── qlproto.h
    └── something.c

```

## Basic idea behind the design

- Each module in the code get's a folder with a single header file which specifies the API of that module.
- Modules can be easily included with `<qaul/qlmodule.h>` from anywhere in the project
- All modules require the `formatting` module as all data structures are defined there
- `networking` will at some point only be an API for plugins in `qaulconnector`
- The global API is exposed in `include/qaul/qaul.h` (or similar to support versioning)
- A single `.c` which implements all API functions but calls functions in API submodules for actual implementation
  - So for example `ql_users_create_user(...)` is "implemented" in `src/api/qaul.c` whereas the actual code implementation is more likely in `src/api/admin/identities.c`
  - This creates modular code which is well organised
- There is also a utility module which is either part of libqaul or just qaul-utils

### Note

To achieve the behaviour that libqaul has a public api, per-module APIs but not generally library public header files we probably need two include directories. One which contains `include/qaul.h` which defines the external library API and one which will most likely be in `src/include/...` which then contains all the header files for each module. This folder should be included in the `PRIVATE` scope in cmake to allow all library internal code to access the headers but nothing from outside the project

spacekookie, 2018-02-15:
This is already being worked on and tracked by #212

closed 2018-02-15 by spacekookie