qaul/qaul.net #211: Implement qaul.net HTML5 GUI using EmberJS

MathJud, 2018-01-04:
The current HTML5 interface is the basic GUI for qaul.net. It is currently implemented with 'Jquery Mobile'. The code is very complex and inflexible regarding all the new features and Tasks.

The HTML5 GUI should be modularized and implemented using the EmberJs framework.
https://emberjs.com

## Modules

- [ ] Messaging
- [ ] User Management
  - [ ] Identity management
  - [ ] Identity generation
  - [ ] User trust management
- [ ] File Sharing
- [ ] Call Handling
- [ ] Preferences
  - [ ] Network interface configuration
  - [ ] Internet sharing
  - [ ] Language configuration
- [ ] Information
  - [ ] Network structure
  - [ ] Application status information
  - [ ] Usage and helping information

dhruvdutt, 2018-01-31:
Could it be done with something more popular like [React.js](https://reactjs.org/)

spacekookie, 2018-01-31:
No 😛 

We have our issues to prefer ember.js over any of the other frameworks out there; stability, community and length of support to name a few.

dhruvdutt, 2018-02-01:
@spacekookie I would be really excited to work on this if it had been React.js

spacekookie, 2018-02-01:
I mean, it is possible to learn a new framework 😉 ember.js isn't *that* complicated

LakshSingla, 2018-02-13:
I am interested in undertaking this project. C I have not yet used Ember.js, yet but I have a general understanding of the framework and am familiar with node.js ecosystem along with some Vue experience. Also what would be a good place for me to start working on this.

MathJud, 2018-03-23:
@LakshSingla
Please check out the repository with the Ember GUI ( https://github.com/qaul/qaul-web ) and compare it with the old HTML5-GUI which is in the GUI folder of this repository ( https://github.com/qaul/qaul.net/tree/qaul-ng/GUI  old GUI HOWTO: https://github.com/qaul/qaul.net/blob/qaul-ng/doc/GUI.md ).
The Ember GUI should use the new REST API of qaul.net

jshreyans551, 2019-02-27:
Hi! 
I am interested in this project. I do not have previous experience with Ember but I'm willing to learn. I havve extensive experience with HTML/CSS and JS. Apart from this I'm familiar with NodeJS and have some experience with the React framework.

> Please check out the repository with the Ember GUI ( https://github.com/qaul/qaul-web ) and compare it with the old HTML5-GUI which is in the GUI folder of this repository

@MathJud I've gone through these and feel I can start working once I'm more familiar with Ember. 

> The Ember GUI should use the new REST API of qaul.net

Could you elaborate on this a bit?

santoshvijapure, 2019-03-06:
hey, @MathJud
I'm interested in this project for GSoC19. I'm a third-year undergraduate from India. I never used EmberJs, previously I have worked with some node angular and expressJs projects but I'm pretty comfortable with node frameworks and RESTful APIs and routing methods.

I can start learning EmberJs and working on this project with your approval. :)

santoshvijapure, 2019-03-06:
@jshreyans551 

> The Ember GUI should use the new REST API of qaul.

It means that the  new EmberJs GUI should use the same RESTful APIs which are created for the currunt [qaul.net](qaul.net) website.

Parth910, 2019-03-11:
Hello @MathJud  @spacekookie 
I am interested in this project for GSoC2019.I have some past experience in EmberJs.I worked in EmberJs Projects. So please give me some guidance about how and from where I can start Developing this project. Also previously I worked in NodeJs and  ReactJs projects.
Thanks.

MathJud, 2019-03-18:
> > The Ember GUI should use the new REST API of qaul.net
> 
> Could you elaborate on this a bit?

@jshreyans551 The REST API is defined here: https://github.com/qaul/qaul.net/tree/master/qaul-rest
It is already predefined but still needs to be developed together with the EmberJs Gui.
There shall be a REST API mockup server in order to ease the development.

MathJud, 2019-03-18:
Welcome @LakshSingla @jshreyans551 @santoshvijapure  @Parth910  to the qaul.net develpment community!!! We are looking forward to answer all of your questions and support your development journeys with qaul.net.

I updated the main note on the top and added a section on how to dive into the subject. 

Looking forward to your comments!

MathJud, 2019-03-20:
**GSoC Decision - This is no GSoC Topic at the Moment**

Dear GSoC interested, after a team meeting, we decided that we will not offer a gsoc project for the ember.js UI. Our rationale behind this decision is that the UI by itself is not complex enough to warrant a project and is connected to more UX design work than code.

On the other hand there is a lot of work to be done in the upper layers of the backend (the http API as well as internal testing).

Because of this we will offer two Rust-oriented projects for GSoC. The webui should not become an afterthought and get the time in UX it deserves. We would still welcome and mentor contributers in this domain, but outside of GSoC.

@LakshSingla @jshreyans551 @santoshvijapure @Parth910

closed 2019-03-26 by spacekookie