qaul/qaul.net #217: files

santhoshnp, 2018-09-21:
I have uploaded the file but where i have to see the uploaded files

MathJud, 2018-09-21:
I guess you are talking about the qaul.net file sharing option, right?
You should see the file under the file tab of the sharing user.
You can browse and download other users files by selecting them in the user tab.
Whenever you add a file to the file sharing, it gets advertised in the network via messaging and all users in the network receive a message about new file and can download/add it to their files.

santhoshnp, 2018-09-21:
first off thank for you reply 
i have tried my level best but there is no file under file tab

santhoshnp, 2018-09-21:
[image]
[image]

MathJud, 2018-09-21:
OK. I see you are using the qaul web interface on a windows computer.
On what system is qaul.net running?

santhoshnp, 2018-09-22:
raspberry pi

closed 2019-03-26 by spacekookie