qaul/qaul.net #237: User Network Simulator

MathJud, 2019-03-26:
Write a testing platform to abstract over the low-level details 
of the routing protocol to provide an easy way to emulate a working network
for the upper layers of `qaul.net`.
Someone working on this project would not only be responsible for writing this
simulator but also work with the other `qaul.net` developers on core library
internals as well as API decisions affecting the routing protocol module.

The simulator would act as a kind of quickcheck property testing framework for
the rest of the application, while needing to be tested via property testing itself.

## Milestones

- [ ] Develop simulator concept
- [ ] Implement simulator
- [ ] Run & test the evolving routing protocol
  - [ ] Extend the simulator to be able to support this development

NoraCodes, 2019-04-04:
Hi, I'd be very interested in working on this for GSoC.

MathJud, 2019-04-07:
Dear @NoraCodes welcome to the project. We are looking forward to your proposal!
Don't hesitate to contact us for any questions.

NoraCodes, 2019-04-07:
@MathJud Thanks a lot! I've been communicating with @spacekookie about this on Mastodon. I can send you my draft if there's a good way to contact you privately.

MathJud, 2019-04-08:
Dear @NoraCodes , @spacekookie is the right person to talk to! When you discuss your draft with her, she'll keep me updated.

closed 2021-12-29 by MathJud