qaul/qaul.net #297: Log libqaul errors to file

MathJud, 2022-02-03:


GildedHonour, 2022-02-28:
working on this

GildedHonour, 2022-03-01:
@MathJud 

`log_rust_error(...)` -->

will it be a helper function which would be called manually by user? 
Or would it be called automatically whenever any rust, probably unhandled, error occurs?

GildedHonour, 2022-03-02:
When an error occurs, the app can't simply send it on its own -- without notifying the user, because there may not be internet because it's a Mesh app. Therefore, the user has to be shown and asked to send an error report by himself. 

Correct? @MathJud

MathJud, 2022-03-02:
Yes, the user needs to send it manually.

Some don't even want to log.
You check the GUI screen, where the user can send the logs via email, when you click on the three dots and select debug. There the user also can select if the application shall log at all ...

This just hits me: we should be able to configure it ... 

On a side note. An error in libqaul is usuall no problem and does not let anything to crash. There are only a few that can panick.

GildedHonour, 2022-03-03:
Turning on/off the logger **dynamically** may not be possible because its state has to be specified during compilation. The same goes for logging levels. 
Doing a research, may be there're some workarounds.

GildedHonour, 2022-03-03:
https://rust-lang-nursery.github.io/rust-cookbook/development_tools/debugging/config_log.html#log-messages-to-a-custom-location

`https://rust-lang-nursery.github.io/rust-cookbook/development_tools/debugging/config_log.html#log-messages-to-a-custom-location`

===>

I guess that might work for tech-savvy users.

GildedHonour, 2022-03-03:
Why can't we have a global variable "is_logger_enabled" which will switch from on to off and vise versa when the user checks or unchecks checkbox? This would require to wrap  every `log::debug` with `if is_logger_enabled()`, but it'd work

MathJud, 2022-03-04:
This is a sketch for a solution

1) Use log4rs for logging.
  * https://docs.rs/log4rs/latest/log4rs/
  * configuration & usage example: https://rust-lang-nursery.github.io/rust-cookbook/development_tools/debugging/config_log.html#log-messages-to-a-custom-location
2) use the configuration 
  * you'll get back a handle for later reconfiguration.
  * https://docs.rs/log4rs/1.0.0/log4rs/config/fn.init_config.html
3) reconfigure to write to file, to stop writing to file
  * you can invoke the set_config() function on the handle to change configuration.
  * https://docs.rs/log4rs/1.0.0/log4rs/struct.Handle.html#method.set_config
4) To log to a file you can use the `rolling_file_appender` feature
  * https://docs.rs/log4rs/latest/log4rs/append/rolling_file/index.html

In the logger module we should have the these public functions (dummy code)

```rs
/// 
pub fn init() {
	// create configuration structure
	
	// start logging
	
	// by default log only errors to std_out
	
	// save handle to state
}

/// configure 
/// 
/// save state locally 
pub fn configure(log_enabled: bool, log_level: enum) {
	// create configuration structure
	
	// get handle to logger from state
	
	// set new configuration to logger
}
```

Now we can reconfigure logging whenever we want

A) after loading the confiruation file
B) when receiving a protobuf message from the GUI to set logging a certain way.

## Questions

* Does it work in parallel with android_logger?
  * logging start for android: https://github.com/qaul/qaul.net/blob/5b3786a607cf2fb6cdba765605c53d341cf458dd/rust/libqaul/src/api/android.rs#L41
  * We could use the android_logger only in the debug build, using `#[cfg(debug_assertions)]`
* If the application crashes, can we find the last error in the log file?
  * this should be tested

GildedHonour, 2022-03-04:
Yeah, but our initial requirement is to be to turn it on and off **dynamically**, as well as change lov levels too. That logger doesn't support that, does it?

GildedHonour, 2022-03-04:
- Default (turned off) (checkbox unchecked) - into a stdout / cli
- Turned on (checkbox unchecked) - into a file

GildedHonour, 2022-03-04:
- how to, if possible at all, intercept and write into a log file the latest error which makes an app crash complitely, and which is the case of a crash

GildedHonour, 2022-03-05:
> Does it work in parallel with android_logger? 

[image]

GildedHonour, 2022-03-05:
Let's use `log4rs`.

This library https://github.com/tokio-rs/tracing has been suggested as well, but it's got 300 issues on github, over 1k of commits and hasn't reached 1.0 yet.

GildedHonour, 2022-03-06:
A draft version is ready

GildedHonour, 2022-03-07:
I figure that either log4rs or android_logger will get activated during building automatically. 

Perhaps also

```
#![not(cfg(target_os = "android"))]
```

will have to be into the files that use `log4rs`

GildedHonour, 2022-03-08:
removed the brach alex_mff_dev1
created a new one - issue_297

To test the logger, I guess, you can do this:

```
    log::info!("testing it....");
```

The logs files will be created in the subdirectory `logs/`

MathJud, 2022-06-20:
fixed by #346

closed 2022-06-20 by MathJud