qaul/qaul.net #298: Create dummy message to check whether libqaul is still alive

MathJud, 2022-02-03:


GildedHonour, 2022-03-04:
Similar to "heartbeat" -- no need for any internal fields in Request and Reply.

GildedHonour, 2022-03-05:
Is there any simple example in the code of how UI requests some data, via protobuffers, from the lib? Which I'll then reuse here.

MathJud, 2022-03-05:
An example on how these messages are handled can be found in the users table:

* Send a message to the GUI:
  * https://github.com/qaul/qaul.net/blob/811ed9c2707e00ab8cbe6017eb07219e163854d7/rust/libqaul/src/router/users.rs#L189
* Processing incoming messages from GUI:
  * https://github.com/qaul/qaul.net/blob/811ed9c2707e00ab8cbe6017eb07219e163854d7/rust/libqaul/src/router/users.rs#L156
* The Protobuf file: 
  * The Protobuf definition file [users.proto](https://github.com/qaul/qaul.net/blob/main/rust/libqaul/src/router/users.proto)
  * The automatically generated rust protobuf file [qaul.rpc.users.rs](https://github.com/qaul/qaul.net/blob/main/rust/libqaul/src/router/qaul.rpc.users.rs)

To send a message, send it to the RPC `send_message` function
https://github.com/qaul/qaul.net/blob/811ed9c2707e00ab8cbe6017eb07219e163854d7/rust/libqaul/src/rpc/mod.rs#L189

The messages are received through `process_received_message` function in RPC
https://github.com/qaul/qaul.net/blob/811ed9c2707e00ab8cbe6017eb07219e163854d7/rust/libqaul/src/rpc/mod.rs#L128

GildedHonour, 2022-03-08:
Questions:

(1) GUI will always send `Request` (initiator), and Libqaul will reply with `Reply` (receiver). Never vise versa. Correct? 

(2) All the examples above are about Libqaul. What about GUI: how to send a message from GUI and process a reply in GUI?

(3) Protobuf messages will be sent in a loop, every N seconds? How, is there a similar example?

(4) How can Libqaul become non alive? In what sense non alive?

MathJud, 2022-03-09:
Answers:

1) yes, the GUI will request something from libqaul and libqaul provides it in the response.
2) To see how this is done, you can have a look at the qaul-cli client. It communicates with libqaul via these protobuf messages. The RPC protobuf message sending and receiving is here: https://github.com/qaul/qaul.net/blob/main/rust/clients/cli/src/rpc.rs
3) The messages are scheduled in libqaul and then polled by the UI. In qaul-cli this is done in the main loop: https://github.com/qaul/qaul.net/blob/811ed9c2707e00ab8cbe6017eb07219e163854d7/rust/clients/cli/src/main.rs#L89
4) The problem was, that libqaul panicked. So it was crashed and totally unavailable. Although the main flutter application was still running.

PR #327 (2022-03-14): Added Debug Heartbeat Message
Libqaul has a new debug module that has a heartbeat message.
The UI can send a heartbeat request and libqaul will answer with a heartbeat reply, ensuring the UI it is still running.

fixes #298
commit de897162 (2022-03-14): Added Debug Heartbeat Message

fixes #298 

Libqaul has a new debug module that has a heartbeat message.
The UI can send a heartbeat request and libqaul will answer with a heartbeat reply, ensuring the UI it is still running.
commit cdbb0fce (2022-03-22): qaul debugging: check heartbeat message

Flutter sends heartbeat message to libqaul to check if it is working.
Fixes #298