qaul/qaul.net #300: android start error

MathJud, 2022-02-07:
dart console:

```
[ +198 ms] --------- beginning of main
           02-07 13:03:39.144 D/OPFD_CTRL_SVC( 1335): notify Package Changed { net.qaul.qaul_app } uid = 10, Type = android.intent.action.PACKAGE_ADDED, isVisiblePkg ? false
[  +15 ms] executing: /home/USERNAME/Android/Sdk/platform-tools/adb -s b37eba81 shell am start -a android.intent.action.RUN -f 0x20000000 --ez enable-background-compilation true --ez enable-dart-profiling true --ez enable-checked-mode true --ez verify-entry-points true --ez start-paused true net.qaul.qaul_app/net.qaul.qaul_app.MainActivity
[  +57 ms] Starting: Intent { act=android.intent.action.RUN flg=0x20000000 cmp=net.qaul.qaul_app/.MainActivity (has extras) }

           Error type 3
           Error: Activity class {net.qaul.qaul_app/net.qaul.qaul_app.MainActivity} does not exist.
[        ] Waiting for observatory port to be available...
```

brenodt, 2022-02-16:
Are you still facing this issue?

MathJud, 2022-02-17:
> Are you still facing this issue?

Nope, Android building is fixed atm.

closed 2022-02-17 by MathJud