qaul/qaul.net #314: BLE Module for Linux with BlueR

MathJud, 2022-02-20:
Create a BLE (Bluetooth Low Energy) module in rust which compatible to the two existing modules for Android & iOS.

Existing android BLE module:
https://github.com/qaul/qaul.net/tree/main/android/blemodule

It shall do the following things:

* Listen for BLE advertisements
  * detect neighbour nodes
* Send BLE advertisement messages in regular intervals.
* send arbitrary binary data to nodes nearby.
* receive arbitrary binary data from nodes nearby.

The module is independent and will communicate via protobuf to the controlling backend library libqaul. 

`BlueR` is the official Bluetooth low energy repository for for rust.
https://github.com/bluez/bluer

Prototypes with partial functionality already exist.

6d7a, 2023-02-03:
I'd be interested to have a shot at this issue. To give me little headstart, could you point out at what point the BLE module is started from the flutter app? And regarding the existing prototypes, what branches should I have a look at? Thanks.

MathJud, 2023-02-04:
@6d7a great!

I'm here for questions and discussions!

There are a few points about compatibility:

## 1) The communication via the protobuf sys messages between libqaul and the blemodules. This is the common API for all system dependent modules.

The module can be found here: `rust/libqaul/src/connections/ble`

The protobuf communication between libqaul and the blemodule is defined in this protobuf file: [`rust/libqaul/src/connections/ble/ble.proto`](https://github.com/qaul/qaul.net/blob/main/rust/libqaul/src/connections/ble/ble.proto).

This protobuf file creates the following rust source code: [`rust/libqaul/src/connections/ble/qaul.sys.ble.rs`](https://github.com/qaul/qaul.net/blob/main/rust/libqaul/src/connections/ble/qaul.sys.ble.rs)

The communication principle is relatively simple, libqaul sends a 'request' message and the module responds with the correlating 'response'.

The program flow is the following:

a) At program startup the blemodule is loaded and libqaul sends a `BleInfoRequest`
  * the blemodule checks the capabilities of the available bluetooth module and sends these information in the `BleInfoResponse` message to libqaul.
b) Libqaul checks if a the bluetooth-device is capable of BLE and sends the `BleStartRequest` message to the ble module.
  * the `BleStartRequest` sends the shortened 16 byte 'small qaulid'. (this will be changed in the next weeks to the new shorter 8 byte 'q8id')
  * the blemodule starts the bluetooth device and returns a `BleStartResult` message with the start result: Success or the reasons of an Error
c) When the blemodule starts the following happens:
  * the GATT service is created
  * the module listenes to advertisements
  * the module sends advertisements
d) If a new device is discovered, the blemodule sends a `BleDeviceDiscovered` message to libqaul.
  * The blemodule needs to have a list with all discovered devices and their small qaulid. As some devices change their bluetooth device address randomly every 15 minutes.
  * The device bluetooth addresses are only known by the blemodule and not by libqaul. libqaul interacts with the blemodule only on the basis of the 'small qaulid'.
e) to send a direct message to another device, libqaul sends a `BleDirectSend` message to the blemodule.
  * the blemodule connects to the related device and sends the message via the GATT service.
  * the blemodule returns a `BleDirectSendResult` message to libqaul
f) When the blemodule receives a message from another device it forwards this message to libqaul via `BleDirectReceived` message. 

## 2) The already existing modules for Android and iOS that are being integrated at the moment.

* the Android module is here: [`android/blemodule`](https://github.com/qaul/qaul.net/tree/main/android/blemodule)
* the iOS module is here: [`iOS/QaulBLE`](https://github.com/qaul/qaul.net/tree/main/iOS/QaulBLE)

We have a custom writable GATT service to send direct messages to a device:

* Service ID `99E91399-80ED-4943-9BCB-39C532A7602`
  * write to `99E91402-80ED-4943-9BCB-39C532A7602` to send a message

## General

At the moment we are fixing some last issues of the Android module on the branch [ble-startup-fixing](https://github.com/qaul/qaul.net/tree/ble-startup-fixing)

We will document the exact behaviour of it for that it can be used as a reference.

The BLE communcation shall be enhanced over the next months. It has been implemented this way for the biggest compatibility (compatible with bluetooth 4 devices). There are more specific BLE-modes in Bluetooth 5 that can increase range and bandwith and have the possibility to broadcast bigger amount of data to all devices nearby.

To have a linux version for developing and testing would be very helpful.

I hope this was helpful and gave you a first overview.
Don't hesitate to contact me with any questions.

6d7a, 2023-02-06:
@MathJud, great! Thanks for the explanation. I'll give it a go and push some WIP PRs along the way. I still haven't wrapped my head around the RPC call chain, so just to recap (using Android):
 
1. The android app is started (The app at `qaul.net/qaul_ui/android/app`, right? Is `qaul.net/android/app` used?)
2. The android app loads libqaul and blemodule AARs
3. The app sets up message channels between the UI and libqaul [here](https://github.com/qaul/qaul.net/blob/e31e1c8a8d87186e47b28ec21ae8d60df4c405b7/qaul_ui/android/app/src/main/kotlin/net/qaul/qaul_app/MainActivity.kt#L122)
4. And this is where I'm stuck right now, because I see that there are JNI methods to set up channels for `sys` communication between the blemodule and libqaul [here](https://github.com/qaul/qaul.net/blob/e31e1c8a8d87186e47b28ec21ae8d60df4c405b7/rust/libqaul/src/api/android.rs#L123), but I don't see that they are called anywhere. I'm assuming that the UI component acts as a proxy in the communication between libqaul and the blemodule.

Thanks again for the help

MathJud, 2023-03-19:
@Kevin

Thanks again for this helpful PR!

The interconnection between libqaul's BLE connectivity-module and the Android BLE module is done in the branch `ble-startup-fixing-rebased-06022023` and will be merged to main as soon as all remaining small problems are ironed out.

It is referenced to this PR #533

I rebased your branch on the `ble-startup-fixing-rebased-06022023` branch and called it `linux_ble` pushed it to the qaul repo: 
https://github.com/qaul/qaul.net/tree/linux_ble

And created the following PR from it: #539

---

I change the following things:

* Changed the BLE module to a library.
* The BLE module is running in an own thread within libqaul: https://github.com/qaul/qaul.net/blob/linux_ble/rust/ble_module/src/lib.rs
* The BLE module is directly called from libqaul here: https://github.com/qaul/qaul.net/blob/c523d15769305a57f2efcb639e5ecefba9940cb6/rust/libqaul/src/rpc/sys.rs#L74

TODO:

The BLE module should send it's messages to libqaul.

* I suggest a callback solution, as we did in Android, here:
  * https://github.com/qaul/qaul.net/blob/linux_ble/qaul_ui/android/blemodule/src/main/java/net/qaul/ble/callback/BleRequestCallback.kt
  * https://github.com/qaul/qaul.net/blob/c523d15769305a57f2efcb639e5ecefba9940cb6/qaul_ui/android/blemodule/src/main/java/net/qaul/ble/core/BleWrapperClass.kt#L154

Please don't hesitate to interact on it!