qaul/qaul.net #328: Qaul how to install rk3568 arm  ubuntu18.04?

shaozhai, 2022-03-15:
Qaul how to install rk3568 arm ubuntu18.04?

MathJud, 2022-03-15:
Interesting Question

At the moment the direct compilation of `libqaul` on ARM does not work, as the protobuf prost crate we are using to create the rust source file automatically on every build is not working on this platform. (this should be fixed in the future!)

You can build `libqaul` on another platform using the raspberry build script:
https://github.com/qaul/qaul.net/blob/main/docs/qaul.net-manual/raspberry.md

You could then transfer liblibqaul.so to the rk3568 machine and build the flutter app. 

Unfortunately I don't have a rk3568 machine yet.
But I'm happy to support you building it and I want to organize an rk4568 machine myself.

Looking forward to your comments and findings.

shaozhai, 2022-03-15:
https://m.alibaba.com/product/1600322450437/android-OS-11-RK3568-USB3-0.html?spm=a2706.7843299.1998817009.5.6c5b3e62o98udM&__detailProductImg=%2F%2Fsc04.alicdn.com%2Fkf%2FH120a8fe178cb407a8f8d8d9ad864a757w.jpg_140x140xz.jpg

MathJud, 2022-03-15:
Do you want to run the GUI or just the server version?

shaozhai, 2022-03-15:
server version