qaul/qaul.net #355: UI Fixes

MathJud, 2022-06-30:
Tablet View

-[ ] Content area is right aligned and moves to the right when widening the window, leaving an growing gap between the tab-bar on the left and the content area.

Chat Messenger

- [ ] Show message sent/received status

brenodt, 2022-07-01:
@MathJud regarding the tablet view, that would also apply to the desktop environment. Do you have any suggestions as to what sort of content could take advantage of the real state when in landscape mode?
We can then put a solution in place for both desktop and tablet.

MathJud, 2022-07-02:
At the moment mainly the chat messenger, here a sketch from the UI development [image]

PR #364 (2022-07-15): [fix] flutter UI issues
Closes #355.
closed 2022-07-15 by MathJud