qaul/qaul.net #420: Memory Consumption

Cinorid, 2022-10-01:
The amount of RAM usage is constantly increasing.
[screen-recorder-sat-oct-01-2022-12-45-58.webm](https://user-images.githubusercontent.com/10947730/193402397-4b866109-637e-4fda-b97c-cf1495b236df.webm)

Platform: Window 10 21H1
Versions: Flutter - v2.0.0-beta.2+744 (https://github.com/qaul/qaul.net/releases/tag/v2.0.0-beta.2%2B744-flutter)

MathJud, 2022-10-01:
Thanks for reporting that.
we need to look into it further.

There are many things kept in memory (routing table, last tasks etc.). Also the data base is keeping stuff in memory and deciding itself when to free it and get it again from disk.

However, it is always good to have a controlling eye on the memory!

Cinorid, 2022-10-01:
@MathJud 
After 30 minutes, the amount of memory used reached 1 GB.
[image]

MathJud, 2022-10-01:
# Bug Tracking

## TODO

- [x] Libqaul: Test Libqaul for Memory Consumption
  - Libaul was tested and no memory increase was found
- [ ] Test Flutter contiues RPC Communication
  - [ ] Users List
  - [ ] Group List
  - [ ] Group Invited
  - [ ] Chat Conversation
  - [ ] Feed List

MathJud, 2022-10-05:
We fixed a memory leak in the FFI, which was only affecting desktop versions.
There is still a memory & CPU increase, but a slower one than before.
We are looking into that at the moment.

MathJud, 2022-10-15:
The latest build should fix all issues.

Please make sure to manually delete the following configuration folder:
AppData\Roaming\qaul in your home directory

fixed by #8dee0a8c865b55c251f640130c4c0aa782f290e5
Download build: https://github.com/qaul/qaul.net/releases/tag/v2.0.0-beta.4%2B782-flutter

closed 2022-10-15 by MathJud