qaul/qaul.net #421: Support for old android devices

Cinorid, 2022-10-02:
Currently qaul.net required at least android API level 26.
is there any chance for making this compatible with older devices?

MathJud, 2022-10-05:
The reason we fixed the version to Android 8 (API level 26), is because of the yet to be activated BLE feature.
There was a significant change in the Android API then. So former versions wouldn't support BLE.
Would this still be helpful?

Our own studies showed the most people are using Android 8 and above.
Do you have different experiences and statistics?

Cinorid, 2022-10-06:
according to these statistics (https://gs.statcounter.com/android-version-market-share/mobile-tablet/iran), Android 7, 6, and 5 together are about 10 percent of Android mobile phones.

Cinorid, 2022-11-08:
I changed the minSdk field to check compatibility with older Android versions.
https://github.com/Cinorid/qaul.net/commit/27a2c54d49dc3ef776abcc981afd82fdfa5038b7

I was able to install and run the app on Android 7.0 devices and connect to other peers using WiFi or internet nodes.
But the app stopped on Android 4.4 and 5.0 devices.

Cinorid, 2022-11-10:
Hi @MathJud 
Any estimation on BLE activation?

MathJud, 2022-11-10:
@Cinorid 
We are working on activating it and hope to have it usable within the next two weeks.

MathJud, 2022-11-10:
> @MathJud I changed the minSdk field to check compatibility with older Android versions. [Cinorid@27a2c54](https://github.com/Cinorid/qaul.net/commit/27a2c54d49dc3ef776abcc981afd82fdfa5038b7)
> 
> I was able to install and run the app on Android 7.0 devices and connect to other peers using WiFi or internet nodes. But the app stopped on Android 4.4 and 5.0 devices.

We will need to go through all our (Flutter) dependencies to check if we have any version restrictions on them, which I guess is the case, according to your test outcome. The GUI framework flutter itself supports Android from version 4.4 .

Cinorid, 2022-11-10:
@MathJud 
> We are working on activating it and hope to have it usable within the next two weeks.

Good news, I'll test it and trying to make it compatible for older devices.