qaul/qaul.net #426: Cannot Create Identity

Cinorid, 2022-10-08:
after new installation, cannot create identity.
[image]

[image]

`config.yaml` content

> node:
>   initialized: 1
>   id: 12D3KooWMxv3pEr29WeKL2ZqoJzDN7NqVXvQUZ5swppLnRQcmax4
>   keys: AyrCk4Mi4qzn2oVWd9m80HLPJzh/5TCaB53G8m+D3t+0fO/glycus68x+b6tBtauEODce4gZj7I0XkYbmwHLDQ==
> lan:
>   active: true
>   listen: /ip4/0.0.0.0/tcp/0
> internet:
>   active: true
>   peers:
>     - /ip4/144.91.74.192/tcp/9229
>   do_listen: false
>   listen: /ip4/0.0.0.0/tcp/0
> user_accounts:
>   - name: test
>     id: 12D3KooWE182fqiGejXJJQ9YWF5SDUWMVXqNEPF52sDovGqCSCcE
>     keys: sIwy5+bQ6n82N7PUs3RyGd8NHXsw8wkjGM5sJijGrZg+MCY8BnmSs9r11WaLOsB2h1hYBn/axuGZko2c12HkXw==
>   - name: test
>     id: 12D3KooWHs9o5mSfyMhgWorexj9XKdqSRb6uShu3Wm2r1yVbuKRt
>     keys: 7giGEkPr06U1mGY7fw7Ql7VbEX9tCiQbqC8COGaVwct3lD9CaPs4VPKNsTwPwHeo+OWQoPCbWigXD2RAtncc6w==
>   - name: test
>     id: 12D3KooWNNBKJ4NBCz2mBekCzJi6MfzcNstHrSYVZvReV55nThBC
>     keys: pcBfYN5qgfJUYC7xkXluv6+huqT9LaaboGnhrDrjG1q6co7WTOOkShnKbad4n7v/qLXgUad4d6QbS2oVaH2CTw==
>   - name: test
>     id: 12D3KooWCMiV88aEnmsBijuGQreaWnh4JE53NK4dcr2G7q2WhTrj
>     keys: KpDyK5rCozl21JrYwWd0zljwL8wQ62Khi7V8/moE0Uglv1nvXhCOCeAiFi++umYAbTAzpS1rxEN+2XGj571XnA==
>   - name: test
>     id: 12D3KooWLZbWo1NFhsALrFCRYZPQyMi5MSKcF3T3Pa7Hj3wmrDoM
>     keys: BpKKGcTezYgYk36gdcJhtBNims1snhUtIi1h35qaoSOfp+8e4Xe41ljegCt0Eb7uYupuhY4i2kMh1Q/gjgM92A==
>   - name: test
>     id: 12D3KooWEEPvrsSaUAtQsZrwVZtdsb7xiYidiV1kkwBdL6jPWSsZ
>     keys: YwGsX6TBmW8zxBhquQ03xwngixEfQQE2nRO+aVszPVpBlq2IsgFtRce04asDOs/uK5HP59Oq1I0+mbibXCek4A==
> debug:
>   log: true
> routing:
>   sending_table_period: 10
>   ping_neighbour_period: 5
>   hop_count_penalty: 10
>   maintain_period_limit: 300

OS: Windows 10 x64 - 21H1

Cinorid, 2022-10-09:
I can confirm that it happens randomly. There is also another bug: If I close and reopen the app, the create profile page show up again.

Cinorid, 2022-10-09:
maybe this is related to network state. this is content of log file:
```
05:14:47 [ERROR] test log to ensure that logging is working
05:14:47 [ERROR] failed to create `InterfaceState`: The requested address is not valid in its context. (os error 10049)
```

MathJud, 2022-10-15:
The latest build should fix all issues.

Please make sure to manually delete the following configuration folder:
AppData\Roaming\qaul in your home directory

fixed by #8dee0a8c865b55c251f640130c4c0aa782f290e5
Download build: https://github.com/qaul/qaul.net/releases/tag/v2.0.0-beta.4%2B782-flutter

closed 2022-10-15 by MathJud