qaul/qaul.net #427: Android Apk size is too large

Cinorid, 2022-10-08:


MathJud, 2022-10-21:
Here an stats of the beta 4 (build 782)

| Binary | size |
|--------|------|
| libqaul.so arm64-v8a | 16 MB |
| libqaul.so armeabi-v7a | 12 MB |
| libqaul.so x86 | 15 MB |
| libqaul.so x86_64 | 16 MB |
| playstore apk download | 16.28 MB |
| sharable unified apk (contains all 4 supported architectures) | 117 MB ||

We are working on optimizing it's size.

Cinorid, 2022-10-21:
It's great.
the previous universal apk file size was about 1GB.

MathJud, 2022-10-29:
the size of 1GB usually results from a testing APK when built yourself. The release versions should be much smaller.

You can now download the latest universal APK from the github release page.
The apk is signed by google and should be directly installable.

closed 2022-10-29 by MathJud