qaul/qaul.net #450: Replace `noise-rust` dependency with others

dastansam, 2022-11-03:
## Issue

Currently, we use [`noise-rust`](https://github.com/sopium/noise-rust) crate for some cryptographic libraries. However, it is not really well-maintained and frequently causes version conflicts with core `rust-libp2p` libraries. Due to this, we had to create a clone of this project in `qaul/noise-rust` and made necessary changes to make it work. 

## Proposal

I think we should remove this crate as dependency and use alternative well-maintained crates. Since this crate is just a wrapper of other libraries in essence, we can just directly use those crates. And as for the `noise-protocol`, `libp2p-noise` crate can be used to replace it.

dastansam, 2022-11-03:
@MathJud let me know what you think about this suggestion

MathJud, 2022-11-03:
@dastansam  I agree with the idea to have only one source of the crypto libraries
and your analysis that `noise-rust` is not well maintained.

The following points are to consider:

A) The reason we are using the noise-rust library is that it actually allows us to have access
to the handshake state. We need access to the handshake state, in order to save it, to sustain a restart of the application, as we are using it for end-to-end message encryption and not transport encryption.
For that we need to have read and write access to most values of the handshake.

B) We are using the noise KK handshake. (`libp2p-noise` hasn't implemented it.)

C) In the future we would like to extend the KK handshake with some qaul-specific extension.

closed 2022-12-02 by dastansam