qaul/qaul.net #502: Linux Snap App only shows blank Window

MathJud, 2022-12-09:
When starting the snap qaul app under linux, the app only shows a blank window.

The reason seems to be the location where flutter tries to save the log files.
When starting the snap from the command line `snap run --shell qaul` the following error is displayed:

```txt
$ qaul
flutter: storing log in directory: /home/sojus/Documents/Logs
[ERROR:flutter/lib/ui/ui_dart_state.cc(198)] Unhandled Exception: FileSystemException: Cannot create file, path = '/home/sojus/Documents/Logs/qaul_log-ProviderException-1670593224821.gzip' (OS Error: Permission denied, errno = 13)
#0      _File.throwIfError (dart:io/file_impl.dart:635)
#1      _File.createSync (dart:io/file_impl.dart:273)
#2      StorageManager._storeCompressedLog (package:logger/src/storage_manager.dart:137)
<asynchronous suspension>
```

MathJud, 2022-12-14:
when running the latest beta 12 build, we still get only a blank window to open.

The following snap error log is written:

```txt
= Seccomp =
Time: Dez 14 23:12:53
Log: auid=1000 uid=1000 gid=1000 ses=4 subj=snap.qaul.qaul (enforce) pid=2094375 comm="qaul" exe="/snap/qaul/x1/bin/qaul" sig=0 arch=c000003e 203(sched_setaffinity) compat=0 ip=0x7f23bc0bdc7f code=0x50000
Syscall: sched_setaffinity
Suggestion:
* add 'process-control' to 'plugs'

= AppArmor =
Time: Dez 14 23:12:53
Log: apparmor="DENIED" operation="open" profile="snap.qaul.qaul" name="/proc/sys/vm/max_map_count" pid=2094375 comm="qaul" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0

```

brenodt, 2023-01-16:
@MathJud Since we merged #513, I'm closing this for now 🎉

closed 2023-01-16 by brenodt

MathJud, 2023-01-17:
fixed by #513