qaul/qaul.net #512: [FOSS] version - qaul – قول

MadTooth, 2023-01-12:
> What is 'FOSS' ? - https://f-droid.org/

### You can make a version of your program without any 'proprietary' file(s)/code ?

### And publish on a project 'F-Droid' !

_Examples: 'Proprietary - Google Play Services'_

1. https://github.com/qaul/qaul.net/blob/a1c459aebaa24f4be53db41ae0aca031dfc11849/android/blemodule/build.gradle#L84
2. https://github.com/qaul/qaul.net/blob/0e820d351d8565d067b7d9723b678abe7b153f67/android/blemodule/src/main/java/net/qaul/ble/core/BleWrapperClass.kt#L24

Recommendation: In future updates, split release to architectures (reduction in size .APK file)

licaon-kter, 2023-09-12:
No news? Still using Google proprietary libs for location and stuff

RasheedAZ, 2023-10-31:
Hi. Is it possible to make a fully free and open source version and submit to F-Droid please?

Also, can this issue be opened?

Thanks.

marek22k, 2024-08-13:
Hi, are there any new news?

Fhiss, 2024-11-22:
I only install programs from F-Droid, as it serves as an additional security check and compliance with the source code for me. The absence of Qaul in F-Droid raises my suspicions, and it may be best not to use this program for now.