qaul/qaul.net #531: libqaul <-> UI protobuf RPC authentication & encryption

MathJud, 2023-02-10:
Create an authentication layer to have authenticated & encrypted communication between the UI and libqaul.

The communication between the GUI and libqaul is done via Protobuf [RPC](https://github.com/qaul/qaul.net/tree/main/rust/libqaul/src/rpc) messages.

The current system shall be extended to allow multiple sessions and multiple users. It shall have a user authentitcation and encrypted communication sessions.

* The authentication should be done via user password or local session key.
* The password shall be optional for the default user.
* The sessions should be transport encrypted via Noise protocol.
* multiple sessions shall be possible per user
* multiple users shall be supported per node

sukhman-sukh, 2024-01-22:
Hey @MathJud ,
I am Sukhman Singh.
I would like to work on this issue if it is not solved yet.
Please assign this issue to me .
Thank you.

MathJud, 2024-01-23:
Thanks a lot for your interest in this issue!
This is a complex one, as it needs to be solved on the rust side (in libqaul & in the rust CLI clients) and on the flutter side (UI) in parallel.
I suggest to first create a written concept in the discussion in order to have a good plan on how to create and implement a solution.