qaul/qaul.net #540: Group Settings Menu not displayed on Desktop

MathJud, 2023-03-21:
On Linux, the `Group Settings` options is not displayed when clicking on the three dots menu in the messenger in a group room.

When clicking on the three dots, `show menu` overlay hint is displayed.

[image]

brenodt, 2023-03-22:
@MathJud Are there any error logs?
I've attempted replicating the bug, both in Linux & MacOS, with the local user being either Admin or not, but it seems to work fine on my machine:
[image]
[image]

MathJud, 2023-03-23:
I can see that this bug does not exist, when one opens an already existing group. 

I could replicate this bug, by creating a new group:

- when creating the first new group, it does open, it does not show the settings.
- when creating a further new group, it does not open this group in the window.

PR #542 (2023-03-28): [#540] Chat Menu not updating
## Description
Fixes #540.
The issue occurred when the user had a direct chat open (which has no menu options, thus no menu to open), and then opened a group chat (new or not).
Closing the current open direct chat prior to opening a group chat would remedy the bug, since the chat screen state was built anew.
Likewise, opening a group chat from an already-opened group chat would remedy the bug, as the menu options are the same.

## Solution
Update the overflow menu options on every chat state change ensures this bug no longer occurs.
As an UX improvement, I've also [... cut]
closed 2023-03-28 by MathJud