qaul/qaul.net #547: Add IPv6 Interface to Swarm

MathJud, 2023-04-11:
Currently qaul can be configured to be connectable either via IPv4 or via IPv6. In order for a libp2p swarm to become dual stack (IPv4 & IPv6), we need to add a second interface to each swarm.

The current interface starts listening to IPv4 and a second interface shall start listening to IPv6.

## Sub Tasks

- [ ] Add a second listener to each libp2p swarm, which starts listening to incoming IPv6 connections.
  - [ ] Add a second listener to the Internet connection. https://github.com/qaul/qaul.net/blob/899edace6dbd3e4e251f27b5498103a622b8afae/rust/libqaul/src/connections/internet.rs#L285
  - [ ] Add a second interface to the LAN connection. https://github.com/qaul/qaul.net/blob/899edace6dbd3e4e251f27b5498103a622b8afae/rust/libqaul/src/connections/lan.rs#L256
- [ ] Make interface listening address configurable. Via the configuration qaul configuration file.
  - https://github.com/qaul/qaul.net/blob/main/rust/libqaul/src/storage/configuration.rs

### Optional Advanced Tasks

- [ ] Write an upgrade procedure, for clients to automatically upgrade to the new configuration file.

## Information

This is a good first issue, which touches quite a few areas of the 
Don't hesitate to ask further questions about it, e.g. in the matrix group chat. We will help you with that.

To have some further information for the fields that are touched here:

* **libp2p** - libqaul uses [rust-libp2p](https://docs.rs/libp2p/latest/libp2p/) crate for the LAN & Internet network connections. 
  * The [listen_on()](https://docs.rs/libp2p/latest/libp2p/trait.Transport.html#tymethod.listen_on) function reference.
* **Configuration** - the qaul configuration is written into a specific configuration file.
  * Documentation: http://qaul.github.io/qaul.net/#/qaul/rust/configuration
  * Internally libqaul uses the crate [config](https://crates.io/crates/config) for writing the configuration file.

## Testing

* Check node information. If it shows you the 
* You can test if the node is manually connectable by adding a static internet node to the

Harshil-Jani, 2023-04-15:
@MathJud In the `configuration.rs` do we need to use two structs for Lan and Internet (Eg : Lan_IPv4 and Lan_IPv6) ? Or there is already support for IPv4 connections through IPv4-mapped IPv6 addresses ?

If there is support already then do we really need two `Swarm::listen_on()` ? It should be possible with just one itself i guess.

MathJud, 2023-04-17:
@Harshil-Jani 

In the configuration file users can define the addresses, the socket is listening on.
This is defined via a [multiaddress](https://github.com/multiformats/multiaddr).

[Definition and explanation of the configuration file](http://qaul.github.io/qaul.net/#/qaul/rust/configuration?id=example-configuration)

```
  # multi address configuring the port the internet module listens on
  # 0.0.0.0 - the module is listening on all addresses.
  # 9229 - the port number 9229 zero means the module listens on port 9229 for incoming connections
  listen: /ip4/0.0.0.0/tcp/9229
```

These addresses define on which address and port a swarm is listening on, with `/ip4/0.0.0.0/tcp/0` is listening on any available IPv4 addresses and a randomly assigned port.

The problem is that a single listener therefore only listens to either IPv4 or IPv6. 
While the node is still able to connect to any address, regardless of IPv4 or IPv6, it can itself only be accessed via either the IP version it is listening on.
In order to become dual-stack and to listen on IPv4 as well as IPv6 we need to create multiple listening interfaces.

IMHO, a good possibility for configuration would be to make the `listen` entry an array.
When starting libqaul one could loop over the array and dynamically create a `listen_on()` interface.

```
  # default settings for listening on all IPv4 and all IPv6 interfaces
  listen: [/ip4/0.0.0.0/tcp/0, /ip6/::/tcp/0]
```

Harshil-Jani, 2023-04-17:
@MathJud I was trying to have it `listen: Vec<String>` and make corressponding changes. But it throwed some error related to `config.yaml`. 

Now, I did a hard reset over all my changes but still unable to get through and run qaul-cli. Is there anything breaking on the repo ? I did complete reset hard and I guess this should not happen due to local changes. 
```rust
running libqaul 2.0.0-beta.14
libqaul data on latest version
thread '<unnamed>' panicked at 'can get a local socket: ParsingError(ParseIntError { kind: InvalidDigit })', libqaul/src/connections/lan.rs:258:39
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

MathJud, 2023-04-22:
fixed by #551

closed 2023-04-22 by MathJud