qaul/qaul.net #548: Record and Send Voice Messages

MathJud, 2023-04-11:
In Chat's the user should be able to send voice messages in direct chats. 

- [ ] Record a voice message
- [ ] Send the voice message as a chat file
- [ ] Make voice message playable from the GUI

sukhman-sukh, 2023-12-19:
I would like to work on this issue.
Please assign me this issue.

MathJud, 2023-12-22:
@sukhman-sukh Thanks a lot for this offer.
I assigned you to the issue.

sukhman-sukh, 2024-01-19:
@MathJud  Hey, I was working on this issue. For recording audio from the user, I am using 
```import 'package: record/record.dart';``` library. 
Here It asks for an audio recording path to start recording.
Which path should I specify to this?

Thank you

sukhman-sukh, 2024-01-20:
@MathJud  Sorry for delay. I was working on the issue and went though code base. While implementing the record feature I am using ```import 'package:record/record.dart';``` which is using ```await record.start(const RecordConfig(), path: 'aFullPath/myFile.m4a');``` to start  recording. 
I want to know what path should I specify.
Is qual storing any messages on user end at any location?

MathJud, 2024-01-23:
@sukhman-sukh
Thanks for your work! The file should be saved locally and then handed over via RPC to libqaul. Libqaul will copy the file 

1) record the file into the local flutter app storage.
2) send the path to the recorded file via to a chat group using the protobuf RPC message `chatfile > SendFileRequest` (see https://github.com/qaul/qaul.net/blob/0da91d6af5f8b99cc77f43f436a0608f3dea32fc/rust/libqaul/src/services/chat/chatfile_rpc.proto#L32 )
  - the file will then be copied by libqaul into the libqaul storage, will be encrypted and sent to the chat group.

MathJud, 2024-01-23:
In order to hand the file over to libqaul for sending:

* There is a function in the `libqaul_worker` to send the file to libqaul (see https://github.com/qaul/qaul.net/blob/0da91d6af5f8b99cc77f43f436a0608f3dea32fc/qaul_ui/packages/qaul_rpc/lib/src/libqaul_worker.dart#L261 )
* the file sharing dialog is using the same function (see https://github.com/qaul/qaul.net/blob/0da91d6af5f8b99cc77f43f436a0608f3dea32fc/qaul_ui/lib/screens/home/tabs/chat/widgets/file_sharing.dart#L54 )

MathJud, 2024-07-15:
The feature is almost ready! Thanks to @sukhman-sukh & @brenodt !
The tests have shown the following discussion points and problems that need to be resolved:

- [ ] #643 
- [ ] #644

MathJud, 2024-11-28:
fixed by #615

closed 2024-11-28 by MathJud