qaul/qaul.net #576: Blocking a user doesn't seem to work as the confirmation keeps opening

Chewie9999, 2023-07-25:
I have tried this on both 2.0.0Beta15 and a previous version on Android 10.
Someone seem to be sending spam (or maybe test messages?) in the public list that I don't care about.

Click on users' icon at the left of message
Click "Block User"
"Do you want to block this user?" is displayed
Press "OK"
Screen goes back to message list, but "Do you want to block this user?" is still displayed/displayed again.

No amount of pressing on "OK" or "Cancel" makes the message go away, only the "X" to close the dialog works.
Let me know if I can provide more info, Thanks.

MathJud, 2023-07-25:
@Chewie Thanks for this report.
At the moment the blocked users still appear in the public chat. We need to work on a more fine grained message selection system with spam filter options.

- [ ] Bug: close block user modal dialog window.
- [ ] Feature: better management of messages in public tab of blocked users.

brenodt, 2024-04-04:
Hi there! The proposed PR #577 fixes the bug documented here, but I do agree that we should add upon it with a feature enhancement. @MathJud let's discuss what to include in the feature and I'll open a new PR for it.