qaul/qaul.net #582: How to use BLE

Cinorid, 2023-08-03:
Hi
I built the application according to the documentation, but I failed to connect two Android devices via Bluetooth and transfer any messages.
Can you provide instructions for testing this.

Thanks

MathJud, 2023-08-03:
It should work out of the box.
Bluetooth needs to be switched on on both devices.
Just start the app on both devices and they should find each other and display the other user within a few seconds.
If the users meet for the first time it will take a bit longer (about 10-20 seconds).

MathJud, 2023-08-04:
I just updated the build and run documentation to represent the new build procedure.

http://qaul.github.io/qaul.net/#/qaul/flutter/android

Cinorid, 2023-08-05:
> It should work out of the box.
> Bluetooth needs to be switched on on both devices.
> Just start the app on both devices and they should find each other and display the other user within a few seconds.
> If the users meet for the first time it will take a bit longer (about 10-20 seconds).
> So far, we tested with devices running Android 13, 12, 11 & 10.

Thanks for your assistance
I'm using these two Android devices
Xiaomi POCO M5S
SAMSUNG Galaxy tab A 8.0

Screenshots related to Poco device:
[image]

[image]

[image]

[image]

Screenshots related to Samsung device:
[image]

[image]

[image]

[image]

Bluetooth and WiFi is enabled and even Bluetooth is paired on both devices. But unfortunately nothing is sent or received.

MathJud, 2023-08-08:
After investigating the matter further, the problem seems to be the following: #586

MathJud, 2023-08-10:
This problem should have been fixed by #586 
Feel free to open this bug again, if the problem stil exists.
The fixed code is now published on android playstore & github release.

closed 2023-08-10 by MathJud