qaul/qaul.net #623: Beta 18 prerelease Testing

MathJud, 2024-04-18:
While testing the beta 18 pre-release the following things appeared:

- Android
    - [ ] The new configuration used the `TCP` address for the Community Node, instead of the new `QUIC` address.
- [ ] Network View
    -  the network was not filtering the results and showed the same result on each filter-option.
    - instead of the shortest path it always took the longest path.

MathJud, 2024-04-18:
The Network View

[image]
[image]
[image]
[image]

MathJud, 2024-04-18:
New Internet Node / Community Node

[image]

brenodt, 2024-04-22:
As discussed, the reason the default node didn't take the new `quic` protocol format was that the build for Android, iOS and Windows had used the library assets from the latest release.

Since v18 is flagged as `pre-release`, it's not considered the latest.

This is circumvented by specifying the tag on the `gh release download` commands, which has been done via: https://github.com/qaul/qaul.net/commit/1f34972902b5b0b724091fcde1059363341c134b

I triggered a new Flutter build using the new configuration:
https://app.circleci.com/pipelines/github/qaul/qaul.net/1401

And we can see it succeeded, as, on new installs, the default community node uses the new protocol:
[image]

closed 2024-06-08 by MathJud